Duet3D/DuetWebControl #166: Alternate axis (U,V & W) show "n/a" for position on dashboard

Imlbrown, 2019-10-10:
Hi Christian, 

Excellent GUI and thanks for your effort!

The latest version v2.0.2 shows "n/a" for my U, V & W axis for their position in the dashboard. Version 2.0.0 does show them correctly, but I am uncertain if v2.0.1 has the same issue. I made no changes from to my configs from 2.0.0, uploaded 2.0.2 and the "n/a" values appeared.

I have X,Y,Z,E,U,V & W defined in my configs.

I am running the latest Duet firmware v2.04RC3 and I have a Duex 5 attached.

Thanks,
Larry

fjfricke, 2019-10-15:
Having the same problem (A, B, C axes) with RRF 3. Would be great if you could correct this 😊
Thanks for the awesome new GUI, too!

chrishamm, 2019-10-17:
Thanks, I could reproduce this problem and will fix it in the next version. In the meantime, you can change to the machine position (by clicking on "Tool Position") to display at least some coordinates for extra axes.

chrishamm, 2019-10-21:
Fixed in v2.0.3.

closed 2019-10-21 by chrishamm