Duet3D/DuetWebControl #168: Feature Request - monitoring V_IN in graph

pipersw, 2019-10-15:
Allow to see voltage Spike and drop while printing

chrishamm, 2019-11-05:
Ideally we could add a new sensor type for vIn to M308 in RRF so that it can be seen under the "Extra" page of DWC. Once it's there, it would be possible to show it in the graph, too.

pipersw, 2020-01-23:
https://github.com/dc42/RepRapFirmware/issues/326

pipersw, 2020-12-02:
From dc42:
"It would also be necessary for DWC to understand that it is displaying units other than temperature, so that it can display an additional Y axis. If a 24V VIN was displayed on the same scale as the temperatures, you wouldn't see the fluctuations."

In DWC, it's possible to add a second graph window for VIN ?