Duet3D/DuetWebControl #170: DWC 2.02 exception if a tool heater has not been defined

dc42, 2019-10-15:
Using the following heater and tool definition in RRF 3 (note, heater 2 is not defined but is used by tool 1):

`; RRF 3`
`M308 S0 Y"thermistor" P"bedtemp" T100000 B3950`
`M308 S1 Y"rtd-max31865" P"spi.cs1"`
`M950 H0 C"bedheat" Q10 T0`
`M950 H1 C"e0heat" T1`
`M308 S10 y"mcutemp" a"Mcu"`
`M308 S11 y"drivers"`
`; Heating models`
`M307 H0 A232.3 C1182.1, D2.4 B0		; heater 0 model, use PID`
`M307 H1 A337.1 C137.6, D5.1 B0		; heater 1 model, use PID`
`; Tool definitions`
`M563 P0 D0 H1                       	; Define tool 0`
`G10 P0 S0 R0                        	; Set tool 0 operating and standby temperatures`
`M563 P1 D1 H2`
`G10 P1 S0 R0                        	; Set tool 1 operating and standby temperatures`

DWC 2.02 fails to update current temperatures. The Chrome debug console shows repeated exceptions:

TypeError: Cannot read property 'state' of undefined
    at ToolsPanel.vue?45aa:3
    at o.Ie [as _l] (vue.runtime.esm.js:2630)
    at o.nv (ToolsPanel.vue?45aa:1)
    at o.t._render (vue.runtime.esm.js:3542)
    at o.r (vue.runtime.esm.js:4060)
    at nr.get (vue.runtime.esm.js:4473)
    at nr.run (vue.runtime.esm.js:4548)
    at Xn (vue.runtime.esm.js:4304)
    at Array.<anonymous> (vue.runtime.esm.js:1980)
    at le (vue.runtime.esm.js:1906)

DWC 1.22.6 works in the same configuration and shows Tool 1 header as "undefined".

chrishamm, 2019-10-17:
Will be fixed in 2.0.3.

chrishamm, 2019-10-21:
Fixed in v2.0.3.

closed 2019-10-21 by chrishamm