Duet3D/DuetWebControl #172: Materials if More then 1 Extruder

tkamsker, 2019-10-28:
Hi i have an tool changer setup with 4 heads, And i created different scripts which work for active Extruder. But when i have loaded for example PLA i can't load PLA on 2nd extruder ? Do i really have to create PLA1 ,2, 3 and 4 ? wouldn't it be "more logical" to have an PLA script for material PLA which is handed at active tool ? 
so i think it is an bug 
thx thomas

akurz42, 2019-10-29:
Same here. I've created PLA_1, PLA_2 and PLA_3. Good news is that this seems to work fine with unload and stuff. Still don't quite see why multiple Extruders shouldnt be able to load the same material.

chrishamm, 2019-11-05:
The main idea behind the spools concept is dealing with individual spools and not with filaments. At some point filament length trackers will be implemented so I can understand why it is still a bit confusing.

For the same reason filaments are mapped uniquely per extruder.

closed 2019-11-05 by chrishamm

tkamsker, 2019-11-06:
I see 
But for my use not a good one . So I might now learn vue to overcame that issue ;)-

chrishamm, 2019-11-06:
Feel free but be aware that a related limitation is also present in the firmware. So you'll have to compile your own firmware binary, too.

It would be really easier to create multiple spools with the same properties. It may be worth changing the web interface to support editing of multiple macro files, though - I'll consider that.

akurz42, 2019-11-06:
I'm also not quite sure if this is the best approach on this level. Some slicers already offer (or plan) filament/spool management, which I believe is where it belongs.

On my printer I would expect a rather simple process that goes like "select tool - load PLA/ABS/PETG/whatever material". Spool management would be a bit over the top. Currently the workflow for loading three filaments for my multi-material setup is selecting the respective tools and then run a different macro for each filament, even though the macros contain the exact same codes.

tkamsker, 2019-11-06:
Hi
Ok point taken for my app I have to touch web interface anyway
There has to be a connection to my cloud service 
Then you can share your printer so you have to tell which material is where and if printer is ready . And if print was successful or not.
Thx for input Thomas