Duet3D/DuetWebControl #173: Process bar way off with multiple extruders

akurz42, 2019-10-29:
I have a Prusa i3-like cartesian setup with 3 extruders with a single hotend (switching, not mixing). All seems to work fine so far, except the process bar. Right now, in a ca. 3 hr print with 3 colors, the process bar says this:

- Processing Body_v1.2_0.15mm_PLA_6100.gcode, 100.0 % complete
- Layer 177 of 220, Filament Usage: 21681.2 mm (0.0 mm remaining)

So it's through 177 of 220 layers but indicates that the project is done 100%. And it does that for quite some time already - at about half of the total layer count it already showed 100%

Possible reason: percentage is based on calculated filament usage but only one of the extruders is considered.

akurz42, 2019-10-29:
This is on version 2.0.4, Firmware 2.04RC4 on a Duet 2 Ethernet, BTW.

chrishamm, 2019-11-05:
It sounds like the filament usage isn't correctly parsed by RRF. Can you provide a sample GCode file?

chrishamm, 2020-01-09:
Closing because of missing feedback - probably a config error.

closed 2020-01-09 by chrishamm