Duet3D/DuetWebControl #174: 2.04 Fan2 slider missing

rsurgiewicz, 2019-10-29:
Hi there,

I've just switched from 1.22 DWC interface and In fan section I have only 
"FAN TOOL", FAN0, FAN1 
FAN2 is missing.

I have a led strip attached to FAN2 and it was really comfy to steer PWM using slider.

tekkydave, 2019-10-30:
I have raised a similar issue on the Duet forum:
https://forum.duet3d.com/topic/12645/dwc-2-0-4-no-slider-for-duex2-fan-4

tekkydave, 2019-10-30:
> Hi there,
> 
> I've just switched from 1.22 DWC interface and In fan section I have only
> "FAN TOOL", FAN0, FAN1
> FAN2 is missing.
> 
> I have a led strip attached to FAN2 and it was really comfy to steer PWM using slider.

I found reverting to 2.0.2 fixes the problem

chrishamm, 2019-11-05:
Thanks, I could figure out why this is the case and I will fix it in 2.0.5.

chrishamm, 2020-01-13:
Fixed in v2.0.5.

closed 2020-01-13 by chrishamm