Duet3D/DuetWebControl #175: Ver 2.0.4 "Auto Sleep" does not work

wtarantul, 2019-11-01:
Updated to version 2.0.4 from version 2.0.0-RC3.
 On version 2.0.0-RC3, everything worked. What could be the problem ?

chrishamm, 2019-11-05:
Thanks, I could reproduce this problem and will fix it in v2.0.5.

chrishamm, 2020-01-13:
Fixed in v2.0.5.

closed 2020-01-13 by chrishamm

mmontminy, 2020-04-24:
Was there a regression with Auto Sleep? It hasn't worked for either of my printers for the last several releases. Currently running 2.1.3 on 2 printers with a Duet2 and Maestro. Neither appear to be calling sleep.g when Auto Sleep is selected during a print.

chrishamm, 2020-04-24:
It works as expected both on my Duet 3 and on my Duet WiFi.

mmontminy, 2020-04-24:
So is there a way to debug it then? I haven't been able to get it to work for quite some time.

chrishamm, 2020-04-25:
What firmware are you using? Be aware that the DWC tab needs to remain open and it has to stay connected for this feature to work.

If that is the case, please open the JS console via F12 and copy the full output of

```
JSON.stringify(machineStore
```

once before the print finishes and just after it has finished. You can look at the network requests for rr_gcode requests too, you should find one in there where DWC sends M1 to RRF.

mmontminy, 2020-04-25:
I should have realized that DWC needs to be running for this to work. I'm
willing to bet that's the case. I was thinking it was something happening
in the firmware, not the browser.