Duet3D/DuetWebControl #177: need to reload the page after a bit inactivity

KrohmKoala, 2019-11-15:
Hi, i found that if i let the page open for some time (sometimes some hours ^^) or if i shutdown chrome and relaunch it, i always need to relaunch the page.. didn't before the 2.04

chrishamm, 2019-11-15:
It sounds like DWC wants to re-establish the connection and then the dialog saying "Connecting" never goes away. I've created a related ticket for the used UI component but unfortunately there is no fix available yet: https://github.com/vuetifyjs/vuetify/issues/9588#issuecomment-551078505

KrohmKoala, 2019-11-17:
The fact is that it's not saying connecting or something else, it's just darker

KrohmKoala, 2019-11-17:
[image]
[image]
That's what i have, with how it's normally to compare

ejjenkins, 2019-11-17:
Mine does exactly the same thing.  It is darker and nothing is clickable.  I have to reload the page to make it responsive.

JohnOCFII, 2019-11-17:
I see this too, usually from Mac Safari.  If I clear cookies, it'll work.  Or I can switch to Chrome.

JohnOCFII, 2019-11-24:
> I see this too, usually from Mac Safari. If I clear cookies, it'll work. Or I can switch to Chrome.

I just resolved/found a workaround to this issue in Mac Safari.  I turned on `"Allow Pop-Up Windows"` in `Settings for this Website...` under the `Safari` menu

chrishamm, 2020-01-13:
Version 2.0.5 uses a different "Connecting" dialog so I think this issue has been resolved. I haven't seen this problem since I made that change.

closed 2020-01-13 by chrishamm

mmontminy, 2020-01-18:
I just had this happen again with 2.0.6, so the bug appears to still be there.

chrishamm, 2020-01-21:
What browser and board are you using? Is there a reliable way to reproduce it?

JohnOCFII, 2020-01-21:
I'm no longer having this particular issue, but the connection from DWC to the Duet does seem to need to reconnect every so often.  I don't blame DWC for that.  And I'm using DuetEthernet (wired) to it isn't a WiFi issue either.  But I don't still have this issue of the page going dark, as if there is another window there -- is that what you are still seeing, @mmontminy ?

mmontminy, 2020-01-22:
> What browser and board are you using? Is there a reliable way to reproduce it?

Chrome latest on windows. Board is a Duet2 Wifi with external antenna. I don't generally have connection issues with Duet2, but they do happen occasionally.

When I see it happen, which is relatively often, across different PCs, it's always when the browser tab hasn't been active. ie; I'm doing something in another tab, flip back to the DWC tab to check on the print progress, and everything's grey'd out.

It happens more frequently when the DWC tab hasn't been active for awhile, especially if other tabs have been resource intensive. Once in awhile it will happen within a few seconds of connection. I'll spawn a tab to my printer, while it's loading/connecting, flip to another tab for a minute, come back and it's loaded but grey'd.

mmontminy, 2020-01-25:
2.0.7 might have fixed this. I installed it a few hours ago. The tab's been sitting idle in the background since then. I just selected the tab and got the "connecting..." popup in the corner, which I never saw when I experienced the problem.