Duet3D/DuetWebControl #178: add addional tool and fans

FireflyNJ, 2019-11-15:
I have Duet Web Control 2.0.4 installed, second extruder and two fans are not shown, I cannot find where or how to add additional tools and fans.   They are setup in the config.g

FireflyNJ, 2019-11-18:
still looking on how to get more than one fans to show on DWC.  
but I get second tool to show, I did not have tool 1 in config.g

chrishamm, 2019-11-23:
Tools are now supposed to be defined using the config tool (see https://configtool.reprapfirmware.org) so the corresponding facility has been removed from DWC2. To control the second extruder, create a new tool that has the second extruder assigned. If you do not wish to use a new config bundle, have a look at M563 and edit your config.g (see https://duet3d.dozuki.com/Wiki/Gcode#Section_M563_Define_or_remove_a_tool).

In order to display more fans, check out the "Print Status" page - there you can toggle the visibility of non-thermostatic fans.

closed 2019-11-23 by chrishamm