Duet3D/DuetWebControl #181: Plug-in system

DMCShep, 2019-11-26:
There are various comments in issue tickets about plans for a plug-in system. Since no dedicated issue ticket nor pull request exists for this, I thought I'd make one to help track any work towards it. 👍 

Previous mentions here:
https://github.com/chrishamm/DuetWebControl/issues/97#issuecomment-449631439
> Dec 23, 2018
> "There is no support for plugins yet but it is planned."

Here:
https://github.com/chrishamm/DuetWebControl/issues/107#issuecomment-467770953
> Feb 27, 2019
> "A plugin system is planned but it will take a while longer before I address it. It may make more sense to design that once the Duet 3 is ready."

And here:
https://github.com/chrishamm/DuetWebControl/issues/153#issuecomment-539424747
> Oct 8, 2019
> "I'd like to integrate it as a separate plugin [...] it's on my TODO list."

DMCShep, 2020-01-15:
Cross-quoting from https://github.com/chrishamm/DuetWebControl/issues/153#issuecomment-574742580

> Just a short update: This is planned for DWC 2.1.0.

chrishamm, 2020-01-15:
Don't get this wrong; the plugin system will not be implemented before DWC 2.2.0 or later. It can be expected sometime this year, though.

DMCShep, 2020-01-15:
Oops, misunderstood, deleted prior comment since it was discussing something else

drobertson123, 2020-05-28:
Just want to add some encouragement for the idea of a plugin capability.

I have some experimental ideas I am working with that have no home in the current web control. It would be great if I could develop a plugin that monitored them.

In addition, I think that other systems have really thrived on the new ideas created in a plugin ecosystem. Octoprint has some amazing examples of what can be done.

I appreciate your work and am impressed with what you have created. It is time to let other people have fun also.

chrishamm, 2020-05-28:
See my latest commit... https://github.com/Duet3D/DuetWebControl/commit/033deb21faaa267a5a507f38ab01afbf562970d9

drobertson123, 2020-06-07:
Nice!!! Is this the first step to broad plugin support?

Looking forward to seeing this direction progress.

mydevpeeps, 2020-09-10:
Some thoughts:
1) Option to trigger a start of the Height Map plugin if G29 is in use?
2) Option under Settings -> General -> Plugins to have a "start on boot" function?

chrishamm, 2020-09-10:
Once a plugin is loaded it should be saved in the settings so that it is loaded automatically next time you open DWC. Is that not the case for you? If not, what browser are you using?

mydevpeeps, 2020-09-10:
Interestingly, I'm in the latest EDGE browser, and tried in chrome, and every time I hit F5, refresh, etc. it asks me to login again. I don't think it's storing session data.. I think it might be because I am using localhost:8080 with 'npm run serve' in powershell and then connecting to the printer from there... It might behave differently if I put it on the printer itself and try it.

chrishamm, 2020-10-02:
The plugin interface has been implemented in 3.2-b1 and the issue you pointed out will be fixed in 3.2-b2. We'll try to provide some demo plugins with the final 3.2 release.

I suggest more ideas/comments about the plugin interface should be discussed in new issues or on the Duet3D forum.

closed 2020-10-02 by chrishamm