Duet3D/DuetWebControl #188: can't create new filament when folder filaments misses

JPluess, 2020-01-03:
hi there, this might be an issue for RRF3.0 RC2 i do not know.

i tried to add new filaments but the folder filaments has not been created on sd card. 
i had to remove the card and create it on the computer. now i can create filaments. 

request: if folder is not on the sd card, could you create a new folder with the web interface? 

thanks. best wishes, joel

chrishamm, 2020-01-03:
RRF tries to create the full folder path including missing parent directories. What board and DWC2 version are you using?

JPluess, 2020-01-03:
i have a board duet 2 wifi , its V1.04

bjbeare, 2020-01-09:
I also see this issue.
Board: Duet 3 MB6HC (MB6HC)
DSF Version: 1.1.0.5
Firmware: RepRapFirmware for Duet 3 MB6HC v0.6 or 1.0 3.0beta12 (2019-11-02b1)

chrishamm, 2020-01-10:
@JPluess You can create the filaments directory manually by opening http://yourmachine/rr_mkdir?dir=0:/filaments We'll fix this in the firmware.

@bjbeare DSF comes with the duetsd package that provides filaments directory so there should be no need to create it manually. Please update your Pi and check if it exists or not.

bjbeare, 2020-01-10:
@chrishamm 
You are right. the directory was there and the files were populated.  I am able to view/edit them using the filaments editor... but the load command can't open the directory. Here's what the directory and permissions look like. The directories are owned by uid=1000 (pi) but the files are owned by root.

[root@bjbeare-desktop sd]$ pwd
/media/bjbeare/rootfs/opt/dsf/sd
[root@bjbeare-desktop sd]$ ls -lRn
.:
total 16
drwxr-xr-x 3 1000 1000 4096 Jan  4 13:23 filaments
drwxr-xr-x 2 1000 1000 4096 Jan  4 19:25 gcodes
drwxr-xr-x 2 1000 1000 4096 Sep 23 09:20 macros
drwxr-xr-x 2 1000 1000 4096 Jan 10 07:53 sys
lrwxrwxrwx 1 1000 1000    7 Oct 22 04:00 www -> ../dwc2

./filaments:
total 4
drwxr-xr-x 2 0 0 4096 Jan  4 13:23 PLA

./filaments/PLA:
total 12
-rw-r--r-- 1 0 0    1 Jan  4 14:06 config.g
-rw-r--r-- 1 0 0 1278 Jan  4 16:36 load.g
-rw-r--r-- 1 0 0    1 Jan  4 14:06 unload.g

./gcodes:
total 23180
-rw-r--r-- 1 0 0   197901 Jan  4 17:41  CFFFP_Body1.gcode
-rw-r--r-- 1 0 0 23245026 Jan  5 08:05 'CFFFP_top stand.gcode'
-rw-r--r-- 1 0 0   282813 Jan  4 16:40  cube.gcode

./macros:
total 0

./sys:
total 80
-rw-r--r-- 1 0 0  457 Jan  3 17:59 bed.g
-rw-r--r-- 1 0 0  662 Jan  4 14:26 config-override.g
-rw-r--r-- 1 0 0 3981 Jan  9 18:27 config.g
-rw-r--r-- 1 0 0 3982 Jan  9 16:06 config.g.bak
-rw-r--r-- 1 0 0 3479 Jan  2 17:57 config.json
-rw-r--r-- 1 0 0  210 Jan  3 17:49 deployprobe.g
-rw-r--r-- 1 0 0 1158 Jan  9 18:42 dwc2settings.json
-rw-r--r-- 1 0 0  367 Jan  3 18:34 heightmap.csv
-rw-r--r-- 1 0 0 1115 Jan  2 18:31 homeall.g
-rw-r--r-- 1 0 0  520 Jan  2 17:48 homex.g
-rw-r--r-- 1 0 0  520 Jan  2 17:48 homey.g
-rw-r--r-- 1 0 0  597 Jan  2 17:48 homez.g
-rw-r--r-- 1 0 0  396 Jan  2 17:48 pause.g
-rw-r--r-- 1 0 0  399 Jan  2 17:48 resume.g
-rw-r--r-- 1 0 0  212 Jan  3 17:45 retractprobe.g
-rw-r--r-- 1 0 0  174 Jan  2 17:48 sleep.g
-rw-r--r-- 1 0 0  206 Jan  2 17:48 stop.g
-rw-r--r-- 1 0 0  161 Jan  2 17:48 tfree0.g
-rw-r--r-- 1 0 0  221 Jan  2 17:48 tpost0.g
-rw-r--r-- 1 0 0  165 Jan  2 17:48 tpre0.g

When I try the Load command, I see:

[image]

This seems to be a different issue than reported on the thread. I can open a new issue if you like.

chrishamm, 2020-01-11:
@bjbeare It doesn't matter if those directories are owned by user 'pi' since 'root' has permission to read/write those files anyway. Have you upgraded to DSF 1.2.3.0 yet? The Filaments feature was not working in 1.1.0.5. Also make sure that you do not have a microSD card in your Duet 3.

dc42, 2020-01-12:
This is fixed in RRF 3.1, which creates the full path recursively when creating a folder or renaming a file.

JPluess, 2020-01-12:
Thanks 😊

⁣TypeApp for Android herunterladen ​

bjbeare, 2020-01-13:
@chrishamm 
Thank you. Problem solved on the load/unload. I didn't realize that I shouldn't have the sdcard installed in the duet3.

closed 2020-01-13 by chrishamm