Duet3D/DuetWebControl #189: Filamentsensor

JPluess, 2020-01-03:
Hi there, 
i checked the endstop section if it showed a filament sensor runout too.
i would like to know if i can see the filament sensor name too. In RRF3 there have been names added. maybe those names can be shown in the web interface?
(its a bit messy to connect since i use 2 individual stepper for z)

its not very obvious wich one has activated since they are sorted by indexnumber, is there a chance that endstops can be mapped flexible?

currently i have drive 
0=>X                 e-stop->X
1=>Y                 e-stop->Y
2=>Z1               e-stop->Z1
3=>Extruder      e-stop->Filament
4=>Z2               e-stop->Z2

should i swap endstops filament and z2?

thanks alot

chrishamm, 2020-01-09:
AFAIK when a filament error occurs, the affected extruder is reported as part of the pause message. If that isn't sufficient for you, have a look at triggers (M581/582) and set up your own actions for now.

Information about sensors is not fully provided by the DWC/DSF object model yet but support will be added sometime soon.

chrishamm, 2020-08-26:
The upcoming DWC 3.2 will display if filament is present or not (with RRF > 3)

closed 2020-08-26 by chrishamm