Duet3D/DuetWebControl #192: DWC replicates error responses when M291 is used

dc42, 2020-01-06:
If a macro file uses M291 to display messages, and a subsequent command display an error or warning message, the error message is replicated one additional time per M291 command. Here is a sample macro:

M291 P"Hello 1" S2
M291 P"Hello 2" S2
M291 P"Hello 3" S2
M291 P"Hello 4" S2
M77

and here is the output from DWC 2.0.4:

06/01/2020, 15:21:08 | M98 P"0:/macros/M291 error test"Warning: M77: Command is not supported
-- | --
06/01/2020, 15:21:08 | M292Warning: M77: Command is not supported
06/01/2020, 15:21:08 | M292Warning: M77: Command is not supported
06/01/2020, 15:21:08 | M292Warning: M77: Command is not supported
06/01/2020, 15:21:08 | M292Warning: M77: Command is not supported

chrishamm, 2020-01-09:
This issue is more related to RRF than it is to DWC. Whenever DWC sends a G-code, it expects 'seq' to be incremented after a sent HTTP code has been processed. As a consequence, DWC then queries rr_reply and resolves pending code tasks (JS Promises) with the given reply.

I think when M292 is executed 'seq' is never touched and the M292 tasks do not complete until the whole macro does. In this case all the pending codes are resolved with the same reply. IMO the best solution for this would be to increase 'seq' whenever RRF processes M292.

chrishamm, 2020-04-03:
Resolved by changing M292 in the firmware, closing.

closed 2020-04-03 by chrishamm