Duet3D/DuetWebControl #194: DWC sometimes uploads firmware files to wrong subfolder

dc42, 2020-01-06:
If a on the System page of DWC goes into a subfolder, any uploads done on the System page go into that subfolder. But this is not correct for firmware files and IAP files; so it can lead to users uploading a new firmware file and then wondering why it doesn't get installed. Binary files should always be uploaded to /sys.

chrishamm, 2020-01-11:
Firmware files are always uploaded to 0:/sys but it isn't the case for IAP files. I'll change DWC2 accordingly.

dc42, 2020-01-12:
I suggest DWC upload all .bin files to /sys, as i can't think of a reason why a .bin file would ever go anywhere else.

chrishamm, 2020-01-13:
DWC 2.0.5 uploads IAP files of the current board to 0:/sys. Since users uploading firmware files from a sub-directory of /sys is a rather rare scenario, I think this solution will be sufficient.

closed 2020-01-13 by chrishamm