Duet3D/DuetWebControl #203: Long pressing issue - mobile

jaysuk, 2020-01-15:
Long pressing on a gcode file whilst in mobile view cause the page to jump to the top.
The usual menu found in the online version does not appear.

chrishamm, 2020-01-15:
Works for me:

[image]

Can you provide more details? What browser are you using?

jaysuk, 2020-01-15:
It works correctly on my machine running RRF 3 stable, DWC 2.0.6 on a duet 2 ethernet.

It doesn't work on my other machine running RRF 3 stable, DWC 2.0.6 on a duet 3 V1 with raspberry pi attached.

chrishamm, 2020-01-15:
No issues on my Duet 3 + RPi either:

[image]

I'll try to reproduce it with Chrome tomorrow.

chrishamm, 2020-01-16:
I could reproduce it in Chrome on Android so I'll check how I can fix this. Interestingly enough, the menu still pops up after the page jumps to the top.

jaysuk, 2020-01-16:
That was quick, I was just replying.

chrishamm, 2020-01-24:
Fixed in v2.0.7.

closed 2020-01-24 by chrishamm