Duet3D/DuetWebControl #204: 2.06 - Temperature Dropdown not displayed when auto-complete disabled

JohnOCFII, 2020-01-18:
Greetings,

In version 2.06 if `Disable Auto-Completion` is active, the temperature drop-down is not displayed, and only manual entry of temperatures is allowed.  Not sure if this is expected behavior.

While I do like the option to turn-off auto-completion on text entry fields, it would be nice if the temperature drop-down would still work.

Test on Macintosh Safari and Chrome.

Thanks.

John

chrishamm, 2020-01-20:
This isn't a bug, it is expected behavior. Auto-Complete means showing drop-down items on inputs so the temperature input is affected as well. Have you tried the auto-complete feature in v2.0.6 yet? IMO it works much better than in previous versions.

JohnOCFII, 2020-01-20:
Thanks for the info.  I did just try leaving auto-complete enabled, and it is behaving much better for me now.  Thanks!

closed 2020-01-20 by JohnOCFII