Duet3D/DuetWebControl #205: 2.0.6 No Popup response to M408, M409 maybe more

LittleHobbyShop, 2020-01-19:
Whilst working through Meta commands I found an odd issue with sending  M408 and M409 and maybe others that provide a response. I'm not getting any popup appear and no log to the console when sending these commands. However, if I have the network log running in developer console on Chrome I do get the response as expected. Popup, and log in the console screen.

chrishamm, 2020-01-21:
That works for me. What board do you have?

LittleHobbyShop, 2020-01-26:
Sorry for the delayed response. I have a Duet 2 Wifi running RRF3.0.1.
It's very intermittent and unfortunately I haven't been able to re-create it though I have had the odd occasion when I have had to send a command twice as there was no response the first time.
There is some discussion here https://forum.duet3d.com/topic/13957/dwc-2-0-6-no-response-to-commands-that-expect-one?_=1579999836948
When I had the original problem in really was like turning on and off a switch with the developer console, responses when it was open, no response when it wasn't. Maybe a local browser issue?

chrishamm, 2020-03-31:
It sounds very much like you had two HTTP sessions running from the same machine. It is a known issue that RRF keeps the G-code response only once per IP address so there is nothing I can do about it in DWC. I haven't been able to reproduce this either.

closed 2020-03-31 by chrishamm