Duet3D/DuetWebControl #207: DWC displays wrong height map

dc42, 2020-01-26:
If you load a specific height map using G29 S1, then if you select Height Map in DWC, users expect iot to show the height map that has been loaded. But it always shows heightmap.csv instead.

Suggestion:
1. We add the current height map filename to the object model (full path from SD card root)
2. The SD version of DWC uses rr_model to query the height map name (if the firmware supports rr_model) and uses that height map file. It would also be helpful to display the name of the height map file alongside the height map.

chrishamm, 2020-01-27:
In fact it is already part of the object model but it may be that the file is not automatically reloaded when it is modified (see [here](https://chrishamm.github.io/DuetSoftwareFramework/api/DuetAPI.Machine.Move.html#DuetAPI_Machine_Move_HeightmapFile)). Does RRF change the reported compensation type to "None" when a new height map is being probed?

dc42, 2020-01-28:
Yes it does set it to None at the start of probing.

chrishamm, 2020-03-31:
Will be fixed in DWC 2.1.0. Whenever `move.compensation.file` changes, the heightmap will be reloaded from the given path.

closed 2020-03-31 by chrishamm