Duet3D/DuetWebControl #209: 2.07 Heater values disappear

cdedwards, 2020-01-28:
I had a odd thing happen yesterday after updating. I selected the bed temp from the list as well as the hotend temp I wanted. a couple of minutes later I noticed the bed temp had changed to a lower value, so I brought up the dropdown list to change it back and noticed that the temp I wanted no longer existed in the list.
[image]

chrishamm, 2020-01-28:
Interesting effect - once you delete the value again from the input or enter something else, the selected value shows up again. TBH I am not sure if this behavior is intentional for this type of control.

chrishamm, 2020-03-31:
In fact it doesn't really matter if this value disappears provided it is the selected value. If you want to apply this value again, just click on the heater next to the input.

closed 2020-03-31 by chrishamm