Duet3D/DuetWebControl #218: Stuck in connecting...

bjbeare, 2020-02-21:
pi@duet3:~/DuetWebControl $ npm install
npm WARN npm npm does not support Node.js v10.15.2
npm WARN npm You should probably upgrade to a newer version of node as we
npm WARN npm can't make any promises that npm will work with this version.
npm WARN npm Supported releases of Node.js are the latest release of 4, 6, 7, 8, 9.
npm WARN npm You can find the latest version at https://nodejs.org/

I then connected to:
http://10.0.0.50:8080. The interface came up but is stuck at "connecting...".
Screenshot enclosed.

[image]

bjbeare, 2020-02-21:
I further tried...
   30  npm run build
   56  unzip -d TEST dist/DuetWebControl-SBC.zip 
   57  ln -s $(pwd)/TEST /opt/dsf/dwc2/TEST

Then connected (externally) to http://10.0.0.50:TEST/index.html
The browser showed a blank page.
Screenshot attached.

... I'm more of a C developer... once I get a starting point on how to deploy/test this, I'll catch up on .js pretty quickly.
Thanks!
[image]

chrishamm, 2020-03-31:
I've been able to reproduce this a couple of times and it will be fixed in DWC 2.1.0.

closed 2020-03-31 by chrishamm