Duet3D/DuetWebControl #223: Can't enter text

thucar, 2020-03-30:
Not sure if this is DWC2 or DW2-for-klipper issue.

Just did a fresh install, the web interface comes up and operates as it should. I can see and control my printer just fine.

However, I'm unable to enter text in any of the text fields. For instance when trying to create a new folder, filament, temperature value, send code, etc. 

The text field would go active but I can not enter anything. Just nothing happens when I try to type something.

thucar, 2020-03-30:
It just occurred to me to test copy-pasting text. That indeed works. So for some reason it looks like keyboard input is disabled.

chrishamm, 2020-03-31:
There was a heightmap-related bug in a previous DWC2 versions that could cause this kind of problem but it has been fixed at least for three months. What DWC version are you using?

thucar, 2020-03-31:
Well duh. Should have checked this first. I was following a guide to get DWC installed and it referred to an older version. Looks like it's version 2.0.0-RC5. Updated to the latest and everything works as it should.

Thank you!

closed 2020-03-31 by thucar