Duet3D/DuetWebControl #224: Feature Request: Option to invert Z on heightmap display

jslaker, 2020-04-02:
Minor quality of life request: on printers like the Railcore where Z+ for the bed moves physically down, the heightmap displays relative to the printer's coordinate space and is inverted from the physical layout of the bed as a result. e.g., a point probed at Z-0.1mm displays as a low point on the heightmap, but is an indication that the bed is physically *high* in that position and needs to be lowered as a result. This is somewhat counter intuitive, especially for new users and is something we see a lot of questions about in Railcore support channels. 

Would it be possible to introduce an optional feature that inverts the display of Z on the heightmap so that it would better match what users are physically seeing on the printer? This would help make things more intuitive for users making bed adjustments on printers with this configuration.

wesc23, 2020-04-02:
Agreed.  Had to figure this out when commissioning my new printer.

RomanLyavin, 2020-04-28:
good afternoon, I would also be happy if you implement this. It will be convenient.

closed 2020-08-26 by chrishamm