Duet3D/DuetWebControl #229: Tool fan speed on job status page

khumarahn, 2020-04-09:
Hi, with the 2.1.1 release, the tool fan speed on the status page is always 0. I fixed this by downgrading to 2.0.4 temporarily.

chrishamm, 2020-04-09:
Will be fixed in DSF 2.1.2. At present it displays the actual speed instead of the requested speed so YMMV.

khumarahn, 2020-04-09:
For me it displayed always 0, even if the requested speed was 100% and fan was working.

I will stay on 2.0.4 for some time. Anyway, thanks a lot!

chrishamm, 2020-04-14:
Fixed in v2.1.2.

closed 2020-04-14 by chrishamm

khumarahn, 2020-04-14:
Thanks! I will test this soon.

RIght now, I updated and I see two fans in DWC: fan 0 and fan 2. But fan 2 is disabled: I have a chamber heater on that pin. I do have fan 1, it is enabled and thermostatically controlled, but DWC does not show it.

chrishamm, 2020-04-14:
Yes, that's a known issue, thermostatically controlled fans are not displayed. However their settings will be visible once the full machine model is displayed on DWC.

khumarahn, 2020-04-14:
Oh... What is "full machine model is displayed on DWC"? What is the machine model? Is this some setting I can change?

And can I hide the disabled fan?

chrishamm, 2020-04-14:
If you are not using fan 3, just configure it as a thermostatic fan or don't configure it at all (if you're using RRF3).

RepRapFirmware 3 now supports conditional G-code with lots of object model values and DWC2 uses those values internally to display everything but it cannot be displayed nicely yet. G/M-codes are still the right way to change those values.

khumarahn, 2020-04-14:
I don't have fan 3. I have fan0 (tool fan), fan 1 (enclosure exhaust fan) and a heater on fan 2 port. Thus fan 2 is disabled (repurposed as a heater). But it shows up in the controls. I can't change its setting, this leads to
```
M106 P2 S0.22
Error: Fan 2 is disabled
```

I updated to RRF3 (because I needed a heater on a fan pin). Thanks for the explanation about the object model... I guess maybe I can change the object model for my printer?

chrishamm, 2020-04-14:
Yes you can change the object model for your printer -- using G/M-codes.
And there must be M106 P2/M950 F2 somwhere in your config.g like:

```
M950 F2 C"fan2" Q500                            ; create fan 2 on pin fan2 and set its frequency
M106 P2 C"LEDs" S0 H-1                          ; set fan 2 name and value. Thermostatic control is turned off
```

Without these codes, fan 2 does not show up.

khumarahn, 2020-04-14:
My section is like this, below. Fan 2 shows up, despite `M950 F2 C"nil"` and `M950 H4 C"fan2" T4`, and fan1 does not, but I don't do M950 for it...
```
M308 S4 P"exp.thermistor4" Y"thermistor" T100000 B3090 C0 R4620 ; configure sensor 4 as thermistor on pin e4temp
M950 F2 C"nil"                                     ; free fan2 (why?)
M950 H4 C"fan2" T4                                 ; fan2 is a heater with sensor 4
M143 H4 S85                                        ; set temperature limit for heater 4
M307 H4 F500 A55 C500 D2.00                        ; PID
M141 H4                                            ; chamber is heated by heater 4
M570 H4 P999999 T100                               ; almost disable temperature protection on heater 4
                                                   ;  - because it is a fan and not a heater

; Fans
M950 F0 C"fan0" Q500                               ; create fan 0 on pin fan0 and set its frequency
M106 P0 S0   H-1                                   ; P0 is part cooling: set to off, no thermostatic control
M106 P1 S128 H-1                                   ; P1 is the exhaust fan, on by default
```

khumarahn, 2020-04-14:
Another thing: I have heaters 0,1,2 and 4 (no 3). But DWC shows graphs for all heaters 0,1,2,3,4 (though heater 3 appears only in the legend).