Duet3D/DuetWebControl #232: Define Area for Mesh Compensation still not working?

RayEbb, 2020-04-17:
Hello,

I have a Duet2 wifi mainboard for a few weeks now.
But from DWC 2.1.0 I can't define the Area for mesh compensation anymore.
When I enter the values and reopen it, then all the settings are zero again.

So now I have to change them in the config.g file, restart my board again, then set the temperatures again, before I can do a G29 command again.

chrishamm, 2020-04-20:
Hello, the values in the "Define area for mesh compensation" dialog are only populated if you are running RepRapFirmware 3.01-RC7 or newer. If the values are different in there from what you configured, that only means that they could not be obtained from the Duet.

Consider upgrading if you are still using RepRapFirmware 2.

RayEbb, 2020-04-20:
> Hello, the values in the "Define area for mesh compensation" dialog are only populated if you are running RepRapFirmware 3.01-RC7 or newer. If the values are different in there from what you [...]

Thanks, for the answer. I'm using RRF 3.0 at the moment

chrishamm, 2020-04-22:
Closing because this bug is not present when RRF 3.01 is installed.

closed 2020-04-22 by chrishamm