Duet3D/DuetWebControl #236: PhpStorm told me that ...

Nibbels, 2020-04-19:
Hello,

first: I am still a beginner when it comes to Vue.
I checked out the project to be able to change The UI on my printer. (I probably need that to debug some stupid behaviour when changing tools on a Toolchanger)

Well, I looked over the code and began to understand the component structure. The code seems to be really clean, except some details the linter told me.
Some of them I wanted to show those to you, in case you dont know about them because you have another development software or another linter.  So please ignore my issue here if that all is bogus.

[image]
That seems to fix it but is it right?:
[image]

I dont really understand this yet:
[image]

[image]

[image]

[image]

Greetings

chrishamm, 2020-04-20:
Thanks for the messages. While several messages do not make sense (like `Condition is always false`) because the referenced fields are either state or getter _properties_, I'll fix the others in the next version.

chrishamm, 2020-04-22:
Fixed in DWC 2.1.4.

closed 2020-04-22 by chrishamm