Duet3D/DuetWebControl #237: Hello Christian Hammacher error in translation Spanish language

Keyn11, 2020-04-20:
Hello Christian Hammacher error in translation Spanish language of Web control
put "Stado" and should "Estado" I am Spanish
On file es.js

chrishamm, 2020-04-20:
Hello Heyn11, can you either upload a fixed es.js file or tell me where to correct it?

Keyn11, 2020-04-20:
If you have it here
[es.zip](https://github.com/chrishamm/DuetWebControl/files/4506369/es.zip)
Thanks

Keyn11, 2020-04-20:
[esFinal.zip](https://github.com/chrishamm/DuetWebControl/files/4506402/esFinal.zip)
More Changes in language
Heater Calefactor
Stado Estado
Thema Tema

chrishamm, 2020-04-20:
I'm afraid that is no plain .js file, it is some kind of saved GitHub page that I cannot use.

Keyn11, 2020-04-21:
I can`t send in Github Window the js files, only ZIP and check for me and download is right if you tell you email and send the file.

chrishamm, 2020-04-21:
Please send it to christian@duet3d.com

chrishamm, 2020-04-22:
As much as I'd appreciate an updated translation file, I cannot merge it unless I get the **actual** JavaScript file and not an HTML file. Please send the raw file over or upload it somewhere, else I will close this issue again.

chrishamm, 2020-04-27:
I'm closing this because the file is unusable. Feel free to send me a pull request and I'll merge your updated translations into the next version.

closed 2020-04-27 by chrishamm