Duet3D/DuetWebControl #238: 2.1.4: Can't change Bed Temp During M190

CCS86, 2020-04-23:
Fairly often I forgot to update the bed temp in my slicer and notice once it is preheating. Generally, I just issue the correct temp via DWC and it moves on. However, after updating to 2.1.4, that is no longer working. 

While waiting for the bed temp, when I entered a new temp, DWC acted as if it accepted the command (progress bar under the field), but the temp kept climbing. After clicking away to another field, the original value from gcode popped back, instead of the value I entered.

Nibbels, 2020-04-23:
(I cannot test and therefore not confirm this bug on 2.1.4 rightnow)

Hmm I think I know that from 2.0.7 at least.
But I did not wonder much because the pre-heat is normally a blocking gcode?

In case there is a possible fix that would be nice.

CCS86, 2020-04-23:
I can verify this via manual gcode in DWC as well.

After issuing an M190, DWC wont accept any other control. I can't override the temp. I can't even "turn everything off".

[image]

chrishamm, 2020-04-24:
This problem has nothing to do with DWC2. If this problem is still present in RRF 3.01-RCx, it is an issue in RepRapFirmware and/or DSF so I suggest you open a new thread about it on the [forum](https://forum.duet3d.com). Closing.

closed 2020-04-24 by chrishamm