Duet3D/DuetWebControl #240: Multiple bed heaters

xyyzyyx, 2020-04-28:
Is DWC able or will be able to show and control mutiple bed heaters in it's layout? That would be helpfull for machines with separate heat zones and multiple heatbeds for multiple material (different temp for different heatzones).

chrishamm, 2020-04-28:
Yes, see [M140](https://duet3d.dozuki.com/Wiki/Gcode#Section_M140_Set_Bed_Temperature_Fast_or_Configure_Bed_Heater). Please use the [forum](https://forum.duet3d.com) for further support questions.

closed 2020-04-28 by chrishamm

xyyzyyx, 2020-04-28:
Sorry but you didn't understand me. I know M140 works, I can controll bed heaters via gcode but I cannot see all bed heaters in DWC layout. There is only one bed heater that I can see, 
[image]
here is an example.
In the bed sectrion there should be 4 heaters, but only one is shown. Also only one is in the graph.

chrishamm, 2020-04-28:
You may need RRFv3 to display multiple bed heaters properly but it does work with DWC 2.1.5:

[image]

All configured via
```
M140 P0 H0
M140 P1 H0
M140 P2 H0
```

xyyzyyx, 2020-04-28:
That's a great news, thanks. 
Seems that there won't be any excuses for upgrading to RRFv3 now :-)