Duet3D/DuetWebControl #241: displaying the fan speed in the web interface

RomanLyavin, 2020-04-28:
Good day Chris, I recently updated DWC2 to the latest version (I previously had version 2.0.7) and encountered the fact that the fan speed indicator displays what is set in the current session, and not the actual values of the fan. I understand that the task was to do this, but can I make a choice - because in the current case it is very inconvenient and it is not clear whether the fan rotates and at what speed, many slicers can set the fan rotation from the printing time, as well as separate settings for bridges and attachments. And now at the time of printing, it is just inconvenient, you can not see what is happening with the printer. If you can please make a separate setting so that it can be as before (so that the fan shows really how much the fan rotates) or as now, it shows the set speed.

chrishamm, 2020-05-06:
In one of the next DWC versions we will display the full object model as returned from RepRapFirmware on a separate page. This will make it easier to view the machine configuration including actual fan values.

mydevpeeps, 2020-09-10:
while we are waiting on this to be implemented, would it be possible for someone to provide a workaround to get the value using the console line?

closed 2022-01-29 by chrishamm