Duet3D/DuetWebControl #249: creating a new filament

cdedwards, 2020-05-13:
I updated to the lastest webcontrol the other day.  Tonight I attempted to create a new filament type and found that there is no option to do so.. There is a "Create File", but no Create Filament.  Create File does exactly that by creating a file in the filament directory instead of creating a directory with load/unload/config g files.

cdedwards, 2020-05-13:
not sure what happened. I reloaded the webpage, and now its Create Filament.. weird.

cdedwards, 2020-05-13:
http://prntscr.com/sfox7h

Created my filament, and lo and behold Create File is back.

chrishamm, 2020-05-13:
What boards and firmware version are you using?

cdedwards, 2020-05-13:
RepRapFirmware for Duet 2 WiFi/Ethernet 3.01-RC12 (2020-05-06b1)
Duet WiFi Server Version: 1.23
Duet Web Control 2.1.7

chrishamm, 2020-05-15:
Thanks, this will be fixed in the next version which we are going to release shortly.

closed 2020-05-15 by chrishamm