Duet3D/DuetWebControl #250: Volumetric extrusion support

zlowred, 2020-05-15:
When printing GCode produced by Cura with "Marlin (volumetric)" setting in machine definition (file header among other stuff says `;Filament used: 35878.7mm3`), DWC reads that as linear millimeters and expects the file to consume 35878.7 linear mm of the filament which is 2.4 times more than actual amount consumed for the job. As a result, all estimations based on the filament consumption are 2.4x off.

Not sure if the problem is on DWC side or RRF side though.

LittleHobbyShop, 2020-05-16:
I suspect that value is produced by your slicer and DWC just picks it up as a comment from the gcode file. You'd be able to test that by switching between volumetric and linear and having a look at the outputs. Cura produces a comment like this:
`;Filament used: 11.731m`

chrishamm, 2020-05-16:
To be clear: DWC does not do any G-code file parsing but RRF and DSF do. I haven't used volumetric extrusion yet, but I'll check again what kind of comment Cura generates.

chrishamm, 2020-05-18:
I get the same output as @LittleHobbyShop, no volumetric extrusion comment is generated for me. Either way, I am closing this issue because it is not directly related to DWC. If you wish to see support for volumetric extrusion parsing, please open a new thread on the [forum](https://forum.duet3d.com).

closed 2020-05-18 by chrishamm