Duet3D/DuetWebControl #252: Response too long with M486

LittleHobbyShop, 2020-05-16:
I'm not sure if this is expected behaviour or not but I'm getting response too long warning with M486 on a print with 10 objects. Sometimes the console lists them all, sometimes not.

GCODE File attached.

[CFFFP_Pi Camera Back.gcode.txt](https://github.com/Duet3D/DuetWebControl/files/4638709/CFFFP_Pi.Camera.Back.gcode.txt)

chrishamm, 2020-05-17:
The output should be displayed in any case on the G-code console. Code replies that are longer than three lines are truncated and the message displayed is "See console". Can you post a screenshot if anything else is wrong?

LittleHobbyShop, 2020-05-18:
Seems I jumped the gun. I thought it was behaving strangely by not listing all objects in the console.  On top of that I misinterpreted the 'Response too long' message. It would appear that RRF parses the object list as they are printed and not right from the start so I was trying M486 before all objects had been picked up. DWC seems to be doing exactly as expected.

Below is from a simulation run.

[image]

closed 2020-05-18 by LittleHobbyShop