Duet3D/DuetWebControl #257: Unable to display the notification information of m291

ningjiazun, 2020-06-03:
Duet Web Control 3.1.1：

[image]
[image]
[image]

*****But it can be displayed in dwc1.22.6

chrishamm, 2020-06-03:
This is a known issue with RRFv2 which I am going to look into. Be aware that RRF2 won't be supported by DWC2 much longer, so please consider upgrading to RRFv3.

htrabbit, 2020-06-17:
I have same issue and I am on rrfv3

chrishamm, 2020-06-17:
Please upgrade to RRF/DWC 3.1.1.

htrabbit, 2020-06-18:
I am on 3.1.1 i think
[image]

htrabbit, 2020-06-18:
AH, Firmware 3.1.1 not web control 3.1.1
lol

that fixed me up 
thank-you so much