Duet3D/DuetWebControl #262: Latency causes issue with UI

charames, 2020-06-10:
Hi,
I’ve got a UI communications issue that I think you should be made aware of.  My printer has a very responsive and usable UI (1.22.6) when I’m connected locally on my network (over WiFi).  But, if I connect over the internet I receive frequent disconnects due to timeout errors.  So, I experimented with the Communication Settings, and found it to be much more stable with:
•	Status Update Interval = 800mS
•	Extended Status Update Every 10th response
•	Max number of Ajax retries = 1

This was suitably functional until I updated the UI to version 2.0.7 and started seeing disconnects again (this time with a message: “Connection interrupted, attempting to reconnect”).  I have tried adjusting the Communication Settings since that worked for version 1.22.6, but no values that I have tried have made it any better.  It still works well when connected locally, but the additional latency of the internet makes the connection unusable.

I can click on the “Revert to DWC1” link and it suddenly works again.  My conclusion is that there is something about the 2.0.7 build that is latency sensitive.

Let me know if there is any additional data collection you could benefit from

Thanks,
-Clint

6/10/2020,   4:42:33 PM | M122   === Diagnostics === RepRapFirmware for Duet 2 Maestro version 2.05.1 running   on Duet Maestro 1.0 Board ID: 08DJM-957AU-PL3TJ-6J1FJ-3SD6N-1B8GV Used output   buffers: 3 of 24 (6 max) === RTOS === Static ram: 19804 Dynamic ram: 92056 of   which 0 recycled Exception stack ram used: 256 Never used ram: 18956 Tasks:   NETWORK(ready,764) HEAT(blocked,1272) MAIN(running,4024) IDLE(ready,160)   Owned mutexes: === Platform === Last reset 00:06:43 ago, cause: software Last   software reset at 2020-06-10 16:35, reason: User, spinning module GCodes,   available RAM 18836 bytes (slot 3) Software reset code 0x0003 HFSR 0x00000000   CFSR 0x00000000 ICSR 0x04418000 BFAR 0xe000ed38 SP 0xffffffff Task 0x4e49414d   Error status: 0 Free file entries: 10 SD card 0 detected, interface speed:   15.0MBytes/sec SD card longest block write time: 12.3ms, max retries 0 MCU   temperature: min 33.1, current 33.5, max 33.7 Supply voltage: min 0.0,   current 24.4, max 24.4, under voltage events: 0, over voltage events: 0,   power good: yes Driver 0: standstill, read errors 0, write errors 1, ifcount   235, reads 25647, timeouts 4 Driver 1: standstill, read errors 0, write   errors 1, ifcount 238, reads 25644, timeouts 7 Driver 2: standstill, read   errors 0, write errors 1, ifcount 245, reads 25646, timeouts 5 Driver 3:   standstill, read errors 0, write errors 1, ifcount 188, reads 25646, timeouts   4 Driver 4: standstill, read errors 0, write errors 1, ifcount 188, reads   25645, timeouts 5 Driver 5: standstill, read errors 0, write errors 1,   ifcount 224, reads 25644, timeouts 6 Driver 6: standstill, read errors 0,   write errors 1, ifcount 188, reads 25643, timeouts 7 Date/time: 2020-06-10   16:42:30 Slowest loop: 27.02ms; fastest: 0.05ms I2C nak errors 0, send   timeouts 0, receive timeouts 0, finishTimeouts 0, resets 0 === Move ===   Hiccups: 0, FreeDm: 160, MinFreeDm: 160, MaxWait: 0ms Bed compensation in   use: none, comp offset 0.000 === DDARing === Scheduled moves: 0, completed   moves: 0, StepErrors: 0, LaErrors: 0, Underruns: 0, 0 === Heat === Bed   heaters = 0, chamberHeaters = -1 -1 Heater 0 is on, I-accum = 0.0 Heater 2 is   on, I-accum = 0.0 === GCodes === Segments left: 0 Stack records: 2 allocated,   0 in use Movement lock held by null http is idle in state(s) 0 telnet is idle   in state(s) 0 file is idle in state(s) 0 serial is idle in state(s) 0 aux is   idle in state(s) 0 daemon is idle in state(s) 0 queue is idle in state(s) 0   lcd is idle in state(s) 0 autopause is idle in state(s) 0 Code queue is   empty. === Network === Slowest loop: 84.09ms; fastest: 0.02ms Responder   states: HTTP(0) HTTP(0) HTTP(0) HTTP(0) FTP(0) Telnet(0) Telnet(0) HTTP   sessions: 1 of 8 Interface state 5, link 100Mbps full duplex
-- | --

mydevpeeps, 2020-08-25:
@charames I think the part of your statement that pulls it away from being a DWC issue is:
"My printer has a very responsive and usable UI (1.22.6) when I’m connected locally on my network (over WiFi). But, if I connect over the internet I receive frequent disconnects due to timeout errors."

I am able to connect to my duet wifi over the internet and I have no issues and did not have to reconfigure anything.  I know others who do the same without issue.

By the way, are you using an asus wifi router?

charames, 2020-08-25:
> EXTERNAL MAIL: harames+caf_=c.harames=f5.com@gmail.com<mailto:harames+caf_=c.harames=f5.com@gmail.com>

Yes, I am using an Asus WiFi router.  Are there known issues regarding that?

mydevpeeps, 2020-08-25:
My was having incredibly weird issues with my duet wifi, connectivity wise, until I disabled the "Airtime Fairness" option on the Asus router. It really messes with the duet since it doesn't really have a strong signal. Mine is in the same room as the router and it only has -39dBm signal strength.

mydevpeeps, 2020-08-25:
re-reading your diag I see you are on wifi and the device is on wired.. which is what my situation was (but opposite).. I was on wired and the duet was wireless... try turning that feature off.

charames, 2020-08-25:
My Duet is not Wifi.
However, your comment about Asus made me try something.  I just added my printer’s IP to the DMZ, and it does seem to be better.  So  Firewall rules may be adding sufficiently to the latency to aggravate the timeouts.

Mine was having incredibly weird issues with my duet wifi, connectivity wise, until I disabled the "Airtime Fairness" option on the Asus router. It really messes with the duet since it doesn't really have a strong signal. Mine is in the same room as the router and it only has -39dBm signal strength.