Duet3D/DuetWebControl #264: Wrong Data on "Machine Properties"

hinotori74, 2020-06-21:
HI all,

Post update to version 3.1.1 i see the folowing data on **"Machine Properties"**:

[image]

But the configuration is correct (Active Hidden Characters):

[image]

Thanks.

chrishamm, 2020-06-24:
Are you using DWC 1.x? That version is no longer supported.

hinotori74, 2020-06-24:
Yes sorry i see the option:

[image]

and I assumed it was supported. I love the look and feel of the DWC 1 interface.

Thanks.