Duet3D/DuetWebControl #269: Tool fan slider not updating to match reality

smartavionics, 2020-07-01:
Recently, I have noticed that the DWC tool fan slider does not track the fan. It stays on 0% even when the fan is running (at any speed). I am currently using DWC 3.1.1 but have noticed this in earlier versions too, I can't remember when it stopped working.

Board: Duet WiFi 1.02 or later (duetwifi102)
Firmware: RepRapFirmware for Duet 2 WiFi/Ethernet 2.05 (2019-12-13b1)
Duet WiFi Server Version: 1.23

smartavionics, 2020-07-01:
I forgot to mention that I am using Chrome on Linux.

amx1, 2020-07-04:
I noticed the same issue. I checked all versions going back starting with 3.1.1, Revision 2.0.7 is the latest version without this issue. 

Board / Firmware: klipper with stefan3 extension / latest versions from github

amx1, 2020-07-04:
a quick fix - only briefly tested:

diff --git a/src/store/machine/connector/PollConnector.js b/src/store/machine/connector/PollConnector.js
index 3d2eefc..f00610a 100644
--- a/src/store/machine/connector/PollConnector.js
+++ b/src/store/machine/connector/PollConnector.js
@@ -321,7 +321,7 @@
                quickPatch(newData, {
                        fans: response.params.fanPercent.map((fanPercent, index) => ({
                                rpm: (index < fanRPMs.length) ? fanRPMs[index] : -1,
-                               value: fanPercent / 100
+                               requestedValue: fanPercent / 100
                        })),
                        heat: {
                                heaters: response.temps.state.map((state, sensor) => ({ state: this.convertHeaterState(state
), sensor }), this)

amx1, 2020-07-04:
a quick fix - only briefly tested

[DuetWebControl.patch.txt](https://github.com/Duet3D/DuetWebControl/files/4873106/DuetWebControl.patch.txt)

smartavionics, 2020-07-05:
Hi @amx1 , your fix appears to work OK, thanks a lot!

robthide37, 2020-07-19:
is this fix going to be added?

mydevpeeps, 2020-08-17:
@amx1 can you share the steps to test the fix?

mydevpeeps, 2020-08-17:
@amx1  & @smartavionics : I'm stuck on trying to setup the environment. I keep getting:

`PS D:\localadmin\duet\DuetWebControl> npm audit

                       === npm audit security report ===

                                 Manual Review
             Some vulnerabilities require your attention to resolve

          Visit https://go.npm.me/audit-guide for additional guidance

  High            Remote Code Execution

  Package         serialize-javascript

  Patched in      >=3.1.0

  Dependency of   @vue/cli-service [dev]

  Path            @vue/cli-service > copy-webpack-plugin >
                  serialize-javascript

  More info       https://nodesecurity.io/advisories/1548

found 1 high severity vulnerability in 1693 scanned packages
  1 vulnerability requires manual review. See the full report for details.`

mydevpeeps, 2020-09-10:
I now have a 4-pin pwm noctua as a part fan and i can confirm the slider is either on or off in prod 3.1.1. I compiled the latest dev version with this change in place and I can confirm the slider works now as expected. Thanks so much!!

mydevpeeps, 2020-09-16:
please add to bug fix list in whats new :-P

chrishamm, 2020-09-16:
I'd say it's part of `Fixed some incompatibilities to RRFv2`. RRFv3 and DSF were not affected.

mydevpeeps, 2020-09-16:
I have RRF 3.1.1 and I'm pretty sure I was able to reproduce it and the fix for it.. but I could be wrong (I am sometimes). 
Also, one of the original comments made was "I noticed the same issue. I checked all versions going back starting with 3.1.1, Revision 2.0.7 is the latest version without this issue." which eludes to it being an issue from 2.0.7+ to 3.1.1?  

Anyway, I guess it's not important. It's your baby. I'm just trying to be useful.

chrishamm, 2020-09-16:
The patch from @amx1 was definitely for the legacy adapter which is only used if connected to RRFv2 (or Klipper + web plugin which emulates that).