Duet3D/DuetWebControl #272: Bug: Need to hit enter/return twice when entering GCode into console.

dmeagor, 2020-07-07:
I think this is a glitch in the intellicode, confused me initially when tuning the heaters and so on.

mydevpeeps, 2020-08-25:
**I was able to reproduce this using the following steps:**
  1. Launch new instance of UI
  2. Type M119 into the quick console line at the top of the UI and press enter. It shows the result.
  3. Wait for the previous timeout (default 5 seconds) or don't.. doesn't matter. 
  4. Do not type anything in quick console line at the top of the UI (M119 should still be there) and press enter.  It does not show the result.
  5. Press enter again (with the previous M119 still in the quick console line at the top). It shows the result.

**How I am able to work around this issue:** 
Enter the command at the top (in this example M119) and click the SEND button instead of pressing enter. I can click this button and it will consistently show the result without any delay. 

**What I see happen:**
  On steps 2& 5 pollConnector.js sends a rr_gcode GET request which returns 200 with the response body of {"buff": 250}. Shortly after rr_reply comes back with response we are looking for and it's displayed in the UI. 

  On step 4 pollConnector.js sends a rr_gcode GET request which returns 200 with the response body of {"buff": 250}. Shortly after rr_reply comes back with response we are looking for and it's not displayed in the UI.

  When I click in the quick code box to type I can see the auto-completion twistie move up, and then back down when I press enter, even though I've disabled this feature. I believe it's related.

  _After I wrote all of this up, I went back and tried it again, and without fail I could not reproduce it and it worked every time by pressing enter. I power cycled the printer and it's still functioning properly now. I am wondering if clicking on the SEND button somehow reset a DOM event chain that stopped this from happening now._

**Configuration Points:**
  - Running UI on localhost:8080 via rpm run serve on master branch
  - General -> Notifications = close automatically, 5000 ms timeout
  - Appearance -> Disable auto-completion
  - Electronics/Versions:
      Board: Duet 2 WiFi (2WiFi)
      Firmware: RepRapFirmware for Duet 2 WiFi/Ethernet 3.1.1 (2020-05-19b2)
      Duet WiFi Server Version: 1.23
      Duet Web Console: 3.1.1

chrishamm, 2020-09-03:
Thanks for the details. When I tried it first, I was unable to reproduce it - it simply worked every time.
But I am using a older hack to hide the auto-complete list so maybe it creates occasional glitches. I'll try to develop another work-around.

chrishamm, 2020-09-03:
Should be fixed in v3.2 If the problem persists, feel free to create a new issue.

closed 2020-09-03 by chrishamm