Duet3D/DuetWebControl #273: Feature: Numeric GCode values are unsuitable for humans

dmeagor, 2020-07-08:
GCode makes sense simple printer commands like movement, particularly on old/slow control board but makes no sense as an actual scripting language and is being misused for complex config and logic.  

Essentially you're asking us humans to act as a very poor language interpreter/compiler, looking up functions in the guide and then entering the GCODE numbers manually into a file (along with all the mistakes) when we could just use intellisense to find what we're looking for and then compile down to a GCODE.  Even assembly language doesn't use numbers for its commands.

Perhaps a parser like ohm-js could translate the syntax from a .hcode to .gcode language for you.  I think the monaco editor used by VSCode could handle a custom .hcode language to for embedding syntax highlighting and intellisense into the web app.

schneik80, 2020-09-04:
I personally chose this firmware because the simplicity of gcode configuration. I do not want some low-code front end. there is already a great web configurator that can be used to generate config files.  The flexibility of directly writing gcode without needing compilation (like marlin) is a HUGE advantage for adapting Duet and reprap firmware to diverse devices.

dmeagor, 2020-09-04:
No you're right, recompiling firmware for a config changes is a terrible idea.  I chose reprap for the same reason, but typing random numbers into a console is no sane solution to live configuration.  

You can type M916, or you can just type "resume" and autocomplete will offer "ResumePrintAfterPowerFailure".

I don't know about you but I personally have a bunch of one line macros created just because the MCodes are not memorable.  M500, "SaveMachineState" and so on.

The gold standard would be a Powershell style solution, clearly too complex but just substituting M/Gcodes for something human readable is just plain sensible.

JPluess, 2021-02-12:
Hi there, I have a similar idea.
Looking for some seldom used g or m codes I need to open the https://duet3d.dozuki.com/Wiki/Gcode
My ideas
1.	Could there be some user shortcuts, like update firmware panel due, M997 S4, they could be configured in a text file like a config.g and then displayed on the console screen. I would prefer just one line with the 5 top commands used, and the rest can be displayed if that line is expanded. Similar like the row below where the past gcode events are shown.
2.	To make life easier, could there be a gcode doku in the DuetWebControl. If I type in “wifi” the gcode list gets filtered accordingly. Hovering over them will display some information maybe an iframe that pulls the info directly form the web.
thanks

mydevpeeps, 2021-02-12:
> 1. Could there be some user shortcuts, like update firmware panel due, M997 S4, they could be configured in a text file like a config.g and then displayed on the console screen. I would prefer just [...]

I have a lot of macros and short of organizing them into folders the list on the right does get quite long. Perhaps we could have the option to mark one as a "favorite" and then have up to <max_defined_in_config> or w/e the visual space allows for the responsive design to still work. Or, we could simply have the favorites show up at the top of the list.

> 2. To make life easier, could there be a gcode doku in the DuetWebControl. If I type in “wifi” the gcode list gets filtered accordingly. Hovering over them will display some information maybe an iframe that pulls the info directly form the web.

Love this idea. Some games I play have this where you can type "wiki <game term>" and it launches the wiki for the game with a search on that topic. It seems what you are suggesting could be most useful on the console command entry area?