Duet3D/DuetWebControl #275: Decode and show embedded gcode thumbnails

christophmuellerorg, 2020-07-24:
PrusaSlicer and SuperSlicer are capable of embedding a thumbnail of the printed object in form of a comment, containing a base64 encoded PNG image. The feature can be enabled by adding a line " thumbnails = 16x16,220x124" to the printers .ini configuration file. It would be great to render these embedded PNG images int the web interface for easier identification of the files.

Attached: 

- sample gcode,containing the thumbnails: [screwtest.gcode.txt](https://github.com/Duet3D/DuetWebControl/files/4974053/screwtest.gcode.txt)
- decoded thumbnails:
[image]
[image]

jltx1, 2021-06-12:
I second this request.  Here's an example of how it is done in OctoPrint plug-in: https://github.com/jneilliii/OctoPrint-PrusaSlicerThumbnails

chrishamm, 2022-05-20:
This is already implemented, closing.