Duet3D/DuetWebControl #276: Bed temperature controls broken depending on which heater is associated with the bed

azvero, 2020-07-30:
I have two RRF2 configurations on the same printer/firmware.  The configurations are identical except for the following:

(A)
M305 S"Bed Plate" P0 X6 R4700 T100000 B3950 
M307 H0 B1 S1

(B)
M140 P0 H7
M305 S"Bed Plate" P7 X6 R4700 T100000 B3950
M307 H7 B1 S1

In (A), everything works as expected (attached figure A).
In (B), bed temperature controls are greyed out in DWC (attached figure B).  This looks wrong.

I would expect temperature controls to work identically in both cases.

[image]
[image]
[image]

n8bot, 2020-07-30:
M305 S"Bed Plate" **P7** X6 R4700 T100000 B3950

Is that not your problem? Shouldn't that be P0?

azvero, 2020-07-30:
I could be misunderstanding the docs, but they say:

Pnnn Heater number (0, 1, 2...) or virtual heater number (100, 101, 102...)

In my case (B) I am trying to use heater 7 for bed 0, which makes me think that P7 is correct.  (Do let me know if you think otherwise.)

https://duet3d.dozuki.com/Wiki/Gcode#Section_M305_Set_temperature_sensor_parameters

azvero, 2020-07-30:
I should also have mentioned that the _physical_ configuration between (A) and (B) is different in that the bed heater is physically connected to H0 and H7 respectively.

n8bot, 2020-07-30:
Yeah, I gathered that you were doing what you're doing. However, it's possible that you still need to put P0. I do not know for sure, but I think the M140 command is kind of like a macro that defines P0 as H7. So, wherever P0 is used, the H7 heater will actually be used but the heater will still be referred to as P0 (if need be.)

I could be entirely incorrect, But I would try that and see if it works. Keep your M140, and maybe even the M307 as H7, but the M305 as P0 and see if it works.