Duet3D/DuetWebControl #279: Jobs List bug

janesler, 2020-08-14:
I am experiencing an issue regarding the Jobs list not showing any of my G-Code. Of the various printers that I own that run RRF 3.1.1 and DWC 3.1.1, a handful of them will fail to bring up the list of Jobs that are stored on the SD card. They only show up when I upload a new file.
[image]

janesler, 2020-08-24:
The machines are running in stand-alone operation. The inspection yielded the following:
[image]

To address what mydevpeeps said, I made sure all of the files in the Jobs list were G-Code only. 
Thanks for your responses.

mydevpeeps, 2020-08-24:
@janesler If I may offer a troubleshooting path in order to provide more details:

1. cycle the power on your printer.
2. open a new tab. open dev tools, network tab selected. 
3. open your dwc url. open the jobs list (expecting it to fail).
4. Upload a file (expecting it to work again as you stated).
5. Hit the refresh button on the jobs page (does the list say or does it disappear?)

In dev tools, click the down arrow button to "Export HAR" file. Attach it here.

janesler, 2020-08-25:
@mydevpeeps I appreciate your help. I've done what you recommended and attached the HAR file (as a .txt file because attaching HAR file is not supported). After hitting the refresh button, the Jobs list failed to open again. 
[Jobs_List-sequence.har.txt](https://github.com/Duet3D/DuetWebControl/files/5124605/Jobs_List-sequence.har.txt)

mydevpeeps, 2020-08-25:
@janesler I reviewed your HAR file and I'm hoping @chrishamm will as well. I see 4 requests go out to rr_filelist (none run rr_fileinfo) and all 4 of them come back as 503 Service Unavailable (confirming what you stated initially). I don't see any rr_upload in your HAR file, so I need you to also provide the HAR file when you perform the upload as well. This is so I can compare the difference between the two requests (after uploading a file it triggers the same rr_filelist and rr_fileinfo requests).

janesler, 2020-08-25:
@mydevpeeps that's strange, I ran an upload of a file named "AA.gcode". Nonetheless, I ran the same process and produced another HAR file. I changed the number of AJAX retries to 5 and the update interval to 500, no success. I also swapped the SD cards on two printers and the problem follows the SD card. Thanks again. 
[Jobs_List-sequence with upload.har.txt](https://github.com/Duet3D/DuetWebControl/files/5126276/Jobs_List-sequence.with.upload.har.txt)

mydevpeeps, 2020-08-26:
@janesler  I see where you have the file upload followed by the directory read, and it seems to work. I will dig into this more thursday when I get more time.  Now that we have determined it's tied to specific SD cards, try something for me (it's an old trick for cell phones).. 

1. Place the SD card into a computer and rename the gcodes directory to DND_BAD_SPACE.  Copy a large file to it (100 MB is probably fine).
2. Create a new directory called gcodes.
3. Put the SD card back into a 3D printer.
4. Upload a gcode file to it.

Restart the printer and see if the listing works. My money is on bad SD cards.  Still, need to be able to accommodate for that somehow in the code, if possible. That is where @chrishamm  comes in.

chrishamm, 2020-08-26:
It is odd that you get a 503 error code for the G-code file list. How many files are in there? In any case I could see from the HAR file that DWC does what it's supposed to, i.e. it requests the G-code reply before it retries the request (long stale G-code replies may prevent other long replies from being sent).

If the problem follows the SD card, you can try to format it once and to check if the problem persists after that. If that doesn't help, throw it away and use another one.

I suggest you post as new thread on the [forum](https://forum.duet3d.com) if you need further support.

closed 2020-08-26 by chrishamm