Duet3D/DuetWebControl #281: Feature Request: M591 sensor status in sensor summary

kiowadriver, 2020-08-16:
I would like to request the Filament Monitor status M591 Dx to be added to the sensor summary on the DWC.  This would make it easier to see the running performance of the sensor as well as determine if filament is loaded when remotely operating the machine.

joaobarros, 2020-08-17:
I thought about this the night before your post!
I second this feature, would be quite handy, I'm currently running a macro and it's not as friendly

kiowadriver, 2020-08-23:
> Sounds like a good idea to me. I'll add an optional icon next to the
> extruder drive title that will indicate if filament is present or not
> (provided the filament sensor type can output the current state). Will be
> implemented in the upcoming v3.2.

Can you also I display the most recent relevant info from M591?

What I have found is that the current sensors detect movent And not
presence.  You have to include a micro switch for presence.

I would like to see the last known M591 data without having to type it in
the console over and over.  The system deletes the data on a pause or print
complete so it is tricky to catch.

chrishamm, 2020-08-23:
No, I'll just add an icon indicating if filament is present or not. The object model does not include that value for every filament sensor type and it is pointless to display the configuration parameters in the status panel.

You'll be able to browse through the object model and to add your own plugins in DWC 3.2 and later.