Duet3D/DuetWebControl #285: Translation to PT-BR

claytonferraz, 2020-09-01:
How can I contribute with the translation of DuetWebControl into Brazilian Portuguese?

chrishamm, 2020-09-01:
You need to translate this file: https://github.com/Duet3D/DuetWebControl/blob/v3.2-dev/src/i18n/en.js Be aware that it will be changed again shortly, though.

claytonferraz, 2020-09-01:
OK, all right, but either way, it’s already a base and should change a few things only.

chrishamm, 2020-09-03:
3.2.0-beta1 is almost ready and the translations won't be changed any further before I release it. If you decide to provide translations, either
- create a new issue where you provide the resulting file
- make a new pull request
- or send me an email with the translated file attached

closed 2020-09-03 by chrishamm