Duet3D/DuetWebControl #287: connection reset event unhandled in events.js

mydevpeeps, 2020-09-02:
**Browser**: Microsoft Edge 44.19041.423.0/Microsoft EdgeHTML 18.19041
**Branch**: master & dev-3.2

**Issue:** Running 'npm run serve' in edge and then connecting to it either by http://localhost:8080 or http://192.168.1.10:8080 (example local system IP) results in a connection reset event, causing a crash of the web server process.

**Console Output:**
 DONE  Compiled successfully in 20082ms                                                                       9:07:27 AM

  App running at:
  - Local:   http://localhost:8080/
  - Network: http://192.168.1.10:8080/

  Note that the development build is not optimized.
  To create a production build, run npm run build.

events.js:167
      throw er; // Unhandled 'error' event
      ^

Error: read ECONNRESET
    at TCP.onStreamRead (internal/stream_base_commons.js:111:27)
Emitted 'error' event at:
    at emitErrorNT (internal/streams/destroy.js:82:8)
    at emitErrorAndCloseNT (internal/streams/destroy.js:50:3)
    at process._tickCallback (internal/process/next_tick.js:63:19)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! duetwebcontrol@3.1.1 serve: `vue-cli-service serve`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the duetwebcontrol@3.1.1 serve script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

mydevpeeps, 2020-09-02:
**update:** Ironically the brand new edge browser didn't have any issues using localhost or the local ip.. so go figure that. still, we should trap for this condition to keep it from hanging. so I want to _emphasize_ the focus of this report is the unhandled event and NOT using edge itself or not running on the printer. thanks.

chrishamm, 2020-09-02:
This is clearly the wrong repository for reporting this. The dev server is provided by Vue.js / Webpack so if you want to complain about it, I suggest you do it there.

I haven't had any issues running the dev server on Linux so far.

closed 2020-09-02 by chrishamm