Duet3D/DuetWebControl #290: 3.2.0-b1+2 : plugins not loading on startup when settings are saved in local storage

mydevpeeps, 2020-09-16:
**Summary:** As soon as I disabled "save settings in local storage" the problem went away and it's now loading the plugins on refresh.

**Troubleshooting:**
Issue: I start any of the 3 new plugins, refresh, and they are not started still. I have tried EDGE and Chrome. It is installed on the printer directly.

I saw that /sys/dwc-plugins.json is empty even after I started one. I remembered I had added that file in to prevent it from crashing when testing the beta before, so I removed the file and refreshed the browser, and then started a plugin again. The file doesn't seem to be getting created.  

_That caused me to start getting 404 errors in the console:_
Failed to load resource: the server responded with a status of 404 (file not found) http://192.168.4.21/rr_download?name=0%3A%2Fsys%2Fdwc-plugins.json

Failed to load resource: the server responded with a status of 404 (page not found<br>Check that the SD card is mounted and has the correct files in its /www folder) http://192.168.4.21/manifest.json

So I added the file back with the content of [] as you previous instructed and refreshed to try again. Same result.

I do see calling /rr_download?name=0%3A%2Fsys%2Fdwc-plugins.json on startup, it's just that nothing is being put in the file when i start one.

I began to wonder so I disabled "save settings in local storage" under General and it caused this odd crash loop. I closed the tab and opened a new one and now it seems to be working as expected.

It seems you added this to the dwc-settings.json? If so, what is dwc-plugins.json for?
,"enabledPlugins":[],"plugins":{}}}

chrishamm, 2020-09-16:
dwc-plugins.json represents a list of installed plugins on a standalone Duet.

Thanks for reporting this, I'll try to reproduce the issue with local storage soon.

chrishamm, 2020-10-01:
I could reproduce this problem. It will be fixed in DWC 3.2-b2.

closed 2020-10-01 by chrishamm