Duet3D/DuetWebControl #291: 3.2.0-b1+2 : ERR_CONNECTION_RESET When connected from two tabs/locations

mydevpeeps, 2020-09-16:
**Summary:**
Unable to use web interface if connected from two sessions at once.

**Troubleshooting:**
I am trying to use the latest Chrome here and now all of a sudden I am getting all of these ERR_CONNECTION_RESET on random file requests.. The most telling error seems to be this one:

index.js:54 Uncaught (in promise) Error: Host 192.168.4.21 is already connected!

This last time it connected but failed to load dwc-settings.json and so my interface was now white and none of my preferences saved.. 

Which files get this error seem to be completely random:

[image]

[image]

[image]

As soon as I closed out my second session (from EDGE) all the issues went away.

chrishamm, 2020-09-16:
Two sessions to a RRF controller in standalone mode isn't really supported at the moment anyway - you'll get race conditions when you send commands from the code console. I haven't seen this particular error yet, though.

mydevpeeps, 2020-09-16:
When I run the M122 Diags report, and it says: "HTTP sessions: 1 of 8"  I guess I just assumed it could handle 8 (authenticated) sessions at once. That's my misunderstand I guess. Thanks for letting me know. 

NOTE: This never happened in the previous GA release of DWC. I've had it open 3 or 4 times without issue on different devices at the same time (desktop, laptop, tablet, etc).

mydevpeeps, 2020-09-16:
I thought, maybe this is because I'm doing it from the same computer. So I connected from my tablet and then hit refresh on my desktop and got the same issue on the desktop, as long as my tablet was connected. 

In a bit of irony if I am running it off the duet AND running it in dev mode (localhost:8080) - the issue still occurs.. so you are definitely on to something there about the connect to the duet from more than one place! I had no idea! Should I post an error over to dc in RRF forums?

chrishamm, 2020-09-16:
Sorry, I forgot the most important part in the previous comment - I meant to say two sessions from the same IP address are not supported.

mydevpeeps, 2020-09-16:
I have confirmed I see this issue from a single computer, single connection - with plugins enabled. as soon as i disable them all this issue goes away forever. reproducable.

chrishamm, 2020-10-01:
Connection resets indicate a network problem between DWC and your board - this is not really related to the new plugins feature. I'd attempt some more network troubleshooting and check if you can get a better connection.

However, there is an issue in 3.2-b1 when plugins that were loaded before cannot be loaded on the next DWC load. I'll fix this in 3.2-b2.

closed 2020-10-01 by chrishamm