Duet3D/DuetWebControl #293: Since 3.2 b no connection anymore host is allready connected

Stephan3, 2020-09-21:
Not sure what ecxatly went wrong. Since current 3.2b  

[image]

it keeps requesting Status1-2 but webinterface is not accessible. It works till 3.1.1
Is there something special i have to take care of in the future ?

Thanks
Stephan

mydevpeeps, 2020-09-21:
@Stephan3 can you disable all plugins and try again. That seemed to make the issues go away for me.

Stephan3, 2020-09-21:
@mydevpeeps i saw your ticket. How to do so without access to webif?
It’s not rrf it’s Klipper.

chrishamm, 2020-10-01:
I think the reason is a messed up dwc-settings.json file, try to delete it.

Note that the file paths for DWC settings/cache have been changed from '0:/sys/dwc2-settings.json' and '0:/sys/dwc2-cache.json' to '0:/sys/dwc-settings.json' and '0:/sys/dwc-cache.json'.

closed 2020-10-01 by chrishamm