Duet3D/DuetWebControl #301: Recommended Remote Dev Setup ( Duet3 / SBC )

AYapejian, 2020-10-20:
**Question:** 

What is the recommended way of running this from your localhost ( say a desktop ) and connecting to the Duet3 running on a Pi (SBC setup)?  OOTB I get cross origin resource errors, something like this:

**Related Issues:** #211 

**Duet Web Control Version/Branch:** `3.2-dev`

**Node version:** `12.19.0`

**Command Run:** `yarn serve` --> Open browser and connect to remote Duet3 on Rpi SBC

**Error Seen:** 
```
Access to XMLHttpRequest at 'http://pi-duet3.local.lan/rr_connect?password=&time=2020-10-20T18%3A3%3A26' from origin 'http://localhost:8080' has been blocked by CORS policy: Response to preflight request doesn't pass access control check: No 'Access-Control-Allow-Origin' header is present on the requested resource.
pi-duet3.local.lan/rr_connect?password=&time=2020-10-20T18%3A3%3A26:1 
``` 

**Error Side Effects:** 

File items are missing due to the above error. ( Macros, sys, etc.. ) 
To note the web app does load ( running with `yarn serve` ), lets me put in the hostname, shows the UI however any file related 

**Possible Workarounds:**

I can think of a couple ways around this listed below (all untested so far but will do if needed or I get bored lol) but am wondering what others are doing to get a smooth local dev env working on a remote duet3/sbc.

1. Disable CORS on chrome `<path to chrome cli> --args --disable-web-security --user-data-dir /tmp/DuetWebControl`
2. Git checkout the repo on the pi, run `yarn/npm serve` from the pi, use `localhost` when prompted to attach in the UI
   1. And to develop use VSCode SSH Remote Workspaces to keep everything looking 'local'
3. Could probably augment the `serve` script target to forward requests to the API and have them appear as if they are going to local via proxy rewrites
4. Develop the whole thing in a container and do the routing like # 3 via `iptables` in a linux env and use VSCode remote container workspaces for good dev experience

Thanks

AYapejian, 2020-10-21:
Got bored so tried the easiest option and this works so far, no CORS errors and files show as you would expect.  Maybe helpful for others searching for the same.  

This alias was written running on macOS Catalina 10.15.7 but I'm sure it'll work for others, probably most linux derivatives as well with the path changes, though that is untested.  Used chrome canary here but change the path to wherever the chrome binary is.

```shell
function chrome.dev.no-cors () {
    "/Applications/Google Chrome Canary.app/Contents/MacOS/Google Chrome Canary"       \
    --args --no-first-run --new-window --disable-web-security                           \
    -disk-cache-dir="$(mktemp -d /tmp/chromecanary.XXXXXX)"                            \
    --user-data-dir="/tmp/chrome.dev.no-cors" </dev/null &>/dev/null &
}
```

AYapejian, 2020-10-21:
I think I'd consider this a pretty good workaround.  Basically the above but set as a vscode workspace launch configuration.  

Auto launches browser, ghives you breakpoint setting and console within vscode, remembers your extensions ( for [vue dev tools](https://chrome.google.com/webstore/detail/vuejs-devtools/nhdogjmejiglipccpnnnanhbledajbpd?hl=en) ).  If you want to close this issue to this workaround that'd be cool unless other ideas start flowing in.  

It would be nice to have this all setup in the repo on checkout but I'll leave that up to @chrishamm as I don't want to make assumptions about preferred tooling.

```json
{
	"folders": [
		{
			"path": "../github-forks/Duet3D_DuetWebControl"
		}
	],
	"settings": {},
	"launch": {
		"version": "0.2.0",
		"configurations": [
			{
				"type":        "chrome",
				"request":     "launch",
				"name":        "DuetWebControl Debug (CHROME)",
				"runtimeExecutable": "/Applications/Google Chrome.app/Contents/MacOS/Google Chrome",
				"env": {
					"NODE_ENV": "development",
					"DEBUG": "vue*"
				},
				"runtimeArgs": [
					"--args",
					"--no-first-run",
					"--new-window",
					"--disable-web-security",
				],
				"userDataDir": "/tmp/chrome-profiles_duetwebcontrol_nocors",
				"url":         "http://localhost:8080",
				"webRoot":     "${workspaceFolder}/src",
				"breakOnLoad": true,
				"sourceMapPathOverrides": {
				  "webpack:///src/*": "${webRoot}/*"
				},
				"skipFiles": ["node_modules"],
				"smartStep": true,
			},
			{
				"type":        "chrome",
				"request":     "launch",
				"name":        "DuetWebControl Debug (CHROME CANARY)",
				"runtimeExecutable": "/Applications/Google Chrome Canary.app/Contents/MacOS/Google Chrome Canary",
				"env": {
					"NODE_ENV": "development",
					"DEBUG": "vue*"
				},
				"runtimeArgs": [
					"--args",
					"--no-first-run",
					"--new-window",
					"--disable-web-security",
				],
				"userDataDir": "/tmp/chrome-canary-profiles_duetwebcontrol_nocors",
				"url":         "http://localhost:8080",
				"webRoot":     "${workspaceFolder}/src",
				"breakOnLoad": true,
				"sourceMapPathOverrides": {
					"webpack:///src/*": "${webRoot}/*"
				},
				"skipFiles": ["node_modules"],
				"smartStep": true,
			},
		]
	}
}

```

chrishamm, 2020-10-24:
Or you could just set `UseCors` in `/opt/dsf/conf/http.json` from `false` to `true`. That's what I generally use for DWC development with SBC setups.

I'll add this to the documentation when I prepare the next plugin how-tos.

chrishamm, 2020-12-15:
Starting from the next software version it will be necessary to add `M586 C"*"` to your config.g (in both SBC and standalone mode) to permit CORS.

closed 2020-12-15 by chrishamm