Duet3D/DuetWebControl #303: DWC 2.0.4 on Duet+Duex returning invalid status

HaythamB, 2020-11-27:
Hey there,

Currently updated to the latest stable release for the Duet2 Wifi, and I've noticed there's an error happening in the rr_status?type=2 API

Machine status, usually returned as a string from the utility function "convertStatusLetter" is now only returning the single-character response in PollConnector.js

For example, usually the status would be "idle" or "processing", but is now returning "I" or "P"

This is applicable for all 3 types of rr_status calls (I've tested all 3). Something must've been broken somewhere along the way.

chrishamm, 2020-11-27:
Please upgrade to at least DWC 3.1.1 (preferrably even to 3.2.0-b4). Version 2.0.4 is no longer supported.

HaythamB, 2020-11-27:
Sorry my mistake, misunderstood the implementation.

closed 2020-11-27 by HaythamB