Duet3D/DuetWebControl #306: Version mismatch on startup

tdcox, 2020-12-08:
I upgraded a 3.1.1 installation to 3.2.0-beta4 and am now hitting a version mismatch with DWC as follows:

1. Power on SBC and Duet 3
2. Connect to DWC from remote computer. Settings/General shows V3.2.0-beta4 running
3. Check monitor connected to SBC.  Settings/General shows V3.1.1 running
4. Press F11 and Ctrl-F5 to refresh browser on SBC. Settings/General shows V3.2.0-beta4 running
5. Reboot SBC.
6. Check monitor connected to SBC.  Settings/General shows V3.1.1 running

This seems to be more than just a browser cache issue.

chrishamm, 2020-12-08:
Interesting. Why do you think it is more than a browser cache issue? It sounds just like that.

Please try to add `rm -rf ~/.cache/chromium` to the start of `/usr/bin/launch-dwc` and check if the problem persists. If it helps I'll change the default launch script as well.

tdcox, 2020-12-08:
Performing a hard reload should refresh the cache, so after doing that manually once with ctrl-F5, I would expect that the problem would vanish until the next upgrade where the cache is once again stale. In this case, the cache appears to be persistent across boot sessions, which is odd.

I'm not seeing any `Cache-Control` or `Last-Modified` headers so this may be the underlying problem in that the browser doesn't have enough information to tell if the cache is stale.

Nuking the cache on startup solves the immediate problem, but it would be better to implement an intentional cache management policy associated with each asset so that long running instances heal automatically. Killing the cache every boot will increase startup times unnecessarily and will impact other applications the user may have installed on the SBC.

chrishamm, 2020-12-09:
I don't know why Chromium keeps using an old version of the web interface but DWS does make use of caching. It sends the timestamp of the file to be sent (+ nonce I think) and at least Firefox keeps track of this:

```
HTTP/1.1 200 OK
Date: Wed, 09 Dec 2020 10:43:37 GMT
Content-Type: text/html
Server: Kestrel
Content-Length: 1441
Last-Modified: Tue, 08 Dec 2020 09:48:27 GMT
Accept-Ranges: bytes
ETag: "1d6cd4746e82221"
```

When you reload the web interface Firefox sends back the cached data as part of the request:

```
If-Modified-Since: Tue, 08 Dec 2020 09:48:27 GMT
If-None-Match: "1d6cd4746e82221"
```

If the file hasn't been changed, DWS sends back `304 Not Modified`, else it sends over the new version. So it's rather an issue with Chromium than with DWC/DWS.

chrishamm, 2020-12-09:
PS: I'll try to change the Cache-Control header slightly and check if I can persuade Chromium to reload the web interface after rebooting.

chrishamm, 2020-12-09:
I think the wohle culprit was DWS omitting `must-revalidate` in `Cache-Control`. Once that is added, Chromium checks the cached version after rebooting. This will be fixed in the next DWS version. Thanks!

closed 2020-12-09 by chrishamm